polhenarejos/pico-openpgp #4: gpg --card-edit -> admin -> factory-reset blocks CHV1, CHV2, CHV3 PIN retry counter

rrottmann, 2023-01-08:
When doing a factory-reset using gpg --card-edit -> admin -> factory-reset all PIN counters get blocked. They were 3 3 3 before.
Is there an APDU custom-command needed to reset the variables?

polhenarejos, 2023-01-09:
Cannot reproduce. Did you get an error when did factory-reset command?

To reset the PIN it can be done with `gpg --edit-card > unblock` command using the RC code (`12345678` default)

polhenarejos, 2023-01-15:
In step `10` I get 3 3 3. 
In step `14` PIN retries are 2 3 3.

Are you using the latest firmware?

rrottmann, 2023-01-29:
Yes, the firmware is latest. Could it be related to using Waveshare RP2040 Zero?

polhenarejos, 2023-02-02:
Really strange. I tried hundreds of combinations and I cannot reproduce it. I'll finish the tests and see if they pass on waveshare.

polhenarejos, 2023-02-05:
Can you try with the latest commit? I pushed lots of fixes.
I also pushed a battery set from gnuk. Give it a try to see if some test fails.

polhenarejos, 2024-07-15:
Lots of bugfixes since then.

Please reopen if still happens with latest version.

closed 2024-07-15 by polhenarejos