polhenarejos/pico-openpgp #7: Problems with the newest firmware

BH2VSQ, 2024-01-04:
I've tried to install the 1.10 1.8 1.6 and 1.4 versions of firmware. After I install the firmware above 1.6(include 1.6), the device manager showed me with a card reader and no card. But when I installed version 1.4, I can see a card reader and card. I can create key on this card but I can't move existed key to the card. Hope newer version of firmware will fix that.

polhenarejos, 2024-01-08:
Hi. Which OS?

nimbius, 2024-04-08:
error is being reproduced on a Waveshare Zero card.  

version pico_openpgp_waveshare_rp2040_zero-1.12.uf2 hangs with a red blinking light error.  firmware does not get loaded it seems.

version pico_openpgp_waveshare_rp2040_zero-1.8.nitrokey3.uf2 (patched with the patch tool on the website) loads fine, however opensc shows no card readers and gpg --card-edit reports no cards.

gpg version 

gpg (GnuPG) 2.4.5
libgcrypt 1.10.3-unknown
Copyright (C) 2024 g10 Code GmbH
License GNU GPL-3.0-or-later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Home: /home/cicero/.gnupg
Supported algorithms:
Pubkey: RSA, ELG, DSA, ECDH, ECDSA, EDDSA
Cipher: IDEA, 3DES, CAST5, BLOWFISH, AES, AES192, AES256, TWOFISH,
        CAMELLIA128, CAMELLIA192, CAMELLIA256
Hash: SHA1, RIPEMD160, SHA256, SHA384, SHA512, SHA224
Compression: Uncompressed, ZIP, ZLIB, BZIP2

files are being copied from bash on arch linux. GNU bash, version 5.2.26(1)-release (x86_64-pc-linux-gnu)

nimbius, 2024-04-08:
more information:

device successfully works in Ubuntu 23.10

gpg version gpg --version
gpg (GnuPG) 2.2.40
libgcrypt 1.10.2
Copyright (C) 2022 g10 Code GmbH

opensc-tool -l
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Nitrokey Nitrokey 3 [Pico HSM Interface] (E6625887D3345230) 00 00
1    Yes             SoloKeys Solo 2 [CCID/ICCD Interface] (F14459D8C72E2E5EB2BB91A1C4BCB771) 01 00

gpg --card-edit

Reader ...........: Nitrokey Nitrokey 3 [Pico HSM Interface] (E6625887D3345230) 00 00
Application ID ...: D276000124010304FFFE453636000000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unmanaged S/N range
Serial number ....: 45363600
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: not forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]

[  215.522435] usb 1-6: new full-speed USB device number 3 using xhci_hcd
[  215.857199] usb 1-6: New USB device found, idVendor=1209, idProduct=beee, bcdDevice= 3.c4
[  215.857213] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  215.857219] usb 1-6: Product: Solo 2 Security Key
[  215.857225] usb 1-6: Manufacturer: SoloKeys
[  215.857230] usb 1-6: SerialNumber: F14459D8C72E2E5EB2BB91A1C4BCB771
[  215.892517] hid-generic 0003:1209:BEEE.0005: hiddev1,hidraw4: USB HID v1.11 Device [SoloKeys Solo 2 Security Key] on usb-0000:02:00.0-6/input1
[  367.332163] usb 3-3: USB disconnect, device number 3
[  370.725068] usb 3-3: new full-speed USB device number 4 using xhci_hcd
[  370.879202] usb 3-3: New USB device found, idVendor=20a0, idProduct=42b2, bcdDevice= 3.04
[  370.879216] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  370.879222] usb 3-3: Product: Pico HSM CCID
[  370.879227] usb 3-3: Manufacturer: Pol Henarejos
[  370.879232] usb 3-3: SerialNumber: E6625887D3345230

could this be some issue with the version of the PKCS11 libraries on the system perhaps?  is there more data i can provide to you for debug analysis?  Thanks again for all your hard work.  these pico projects are awesome!

polhenarejos, 2024-04-08:
Thanks!

In Ubuntu 23.10 it works fine, right? Where is it not working? I need the usb logs from the OS where it does not work.

nimbius, 2024-05-16:
in both ubuntu and arch it does not work.  I believe the root-cause is related to something in opensc-tool, as it doesnt detect the card at all.

polhenarejos, 2024-07-15:
Can you check with v2.2 ?

polhenarejos, 2024-07-15:
Some bugfixes are applied in v2.2. 

Please reopen if still persists in v2.2

closed 2024-07-15 by polhenarejos