polhenarejos/pico-openpgp #10: using pico-opengpg in windows

dev-area, 2024-02-18:
Hi
I couldnt find a way to work with the device on windows 
tried gpg, gpg4win
smartcard is not recognized
is it possible?

thanks
liran

polhenarejos, 2024-02-19:
Do you have OpenSC installed? If so, try to `opensc-tool -an` and `pkcs11-tool -L` to see if it is recognized.

dev-area, 2024-02-19:
it is recognized with opensc-tool but i couldnt find a way to make it work on kleopatra (wingpg) or othe gui program

polhenarejos, 2024-02-19:
So, just for the record, with `OpenSC` and `gpg2` it works, right?

dev-area, 2024-02-19:
correct, opensc and gpg is working ok

polhenarejos, 2024-02-21:
I cannot reproduce it. I installed latest version of gpg4win suite and running Kleopatra 3.2.0. My Pico OpenPGP is being recognized, I can generate 3 keys  and change PIN through "Smart card" tab. Is any error you see?
Did you try to run Kleopatra with Administrative privileges?

polhenarejos, 2024-07-15:
Fixed in 11cb855