polhenarejos/pico-openpgp #11: card not found after firmware loaded

nimbius, 2024-04-03:
waveshare zero device loads firmware with no errors.  device plugs in and is not recognized by GPG

dmesg and debug:

[ 2133.470961] usb 1-5: Product: Pico Key
[ 2133.470966] usb 1-5: Manufacturer: Pol Henarejos
[ 2133.470970] usb 1-5: SerialNumber: E6625887D34F5237
[ 2137.594712] usb 1-5: USB disconnect, device number 16
[ 2138.051644] usb 1-5: new full-speed USB device number 17 using xhci_hcd
[ 2138.372468] usb 1-5: New USB device found, idVendor=1050, idProduct=0116, bcdDevice= 4.00
[ 2138.372488] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2138.372495] usb 1-5: Product: Pico Key
[ 2138.372500] usb 1-5: Manufacturer: Pol Henarejos
[ 2138.372506] usb 1-5: SerialNumber: E6625887D34F5237
[cicero@zed ~]$ sudo gpg --card-edit

gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device

polhenarejos, 2024-04-03:
First restart scdaemon
`gpgconf --kill all`

I use VIDPID 1050:0407. Can you test it with these?

nimbius, 2024-04-04:
issue was on my end, seems like im missing scdaemon perhaps.  functions perfectly in ubuntu :)

closed 2024-04-04 by nimbius