polhenarejos/pico-openpgp #13: CMake fails on pico_add_library

sa2kaa, 2024-05-26:
Hi there
CMake  stops with this fault:

CMake Error at pico-keys-sdk/pico_keys_sdk_import.cmake:270 (pico_add_library):
  Unknown CMake command "pico_add_library".
Call Stack (most recent call first):
  CMakeLists.txt:48 (include)

I checked out the repo with --recursive since it seemed to use submodules.

My RP2040 development environment seem to work for other examples although it might have be set up a year ago or so.

Uncorrelated; since the build failed I downloaded a prebuilt binary for a "pico". I then used https://www.picokeys.com to patch the firmware with some other VID/PID but when the board is connected the USB descriptor still identifies it as 0xFEFF:0xFCFD. The result is the same if I use patch_vidpid.sh. I guess needless to say but gpg does not recognize the key with the patched firmware.

polhenarejos, 2024-05-27:
I'd need the entire log and the cmake command you use.

For the second topic, please open another issue for a better tracking. 

Thanks.

sa2kaa, 2024-05-27:
[CMakeError.log](https://github.com/polhenarejos/pico-openpgp/files/15458515/CMakeError.log)
[CMakeOutput.log](https://github.com/polhenarejos/pico-openpgp/files/15458516/CMakeOutput.log)

cmake .. -DPICO_BOARD=pico -DUSB_VID=0x1234 -DUSB_PID=0x5678

PICO_SDK_PATH is set separately by using source:

export PICO_SDK_PATH="/home/olof/pico-pi/pico-sdk/"

polhenarejos, 2024-05-30:
Log files are not so verbose. Paste stdout here of cmake command. `pico_add_library` is provided by Pico SDK so it might be not properly configured. Which branch/tag are you using?

sa2kaa, 2024-05-30:
I made a new recursive checkout of the picopi-sdk and now building with VID/PID set by my choice. I appended the build result as reference.
[pico_openpgp.zip](https://github.com/polhenarejos/pico-openpgp/files/15505632/pico_openpgp.zip)

sa2kaa, 2024-06-08:
This issue can be closed now since it was problem with the picosdk on my machine.

closed 2024-06-08 by polhenarejos