polhenarejos/pico-openpgp #14: Patching VID/PID failed

sa2kaa, 2024-05-27:
Since my build attempt failed (see other issue)  I downloaded a prebuilt binary for a "pico". I then used https://www.picokeys.com/ to patch the firmware with some other VID/PID but when the board is connected the USB descriptor still identifies it as 0xFEFF:0xFCFD. The result is the same if I use patch_vidpid.sh. I guess needless to say but gpg does not recognize the key with the patched firmware.

I added a ZIP-file containing the original binary and the response

[pico_openpgp_pico-2.0.zip](https://github.com/polhenarejos/pico-openpgp/files/15458585/pico_openpgp_pico-2.0.zip)

polhenarejos, 2024-05-30:
I compared both files in the zip and `yubikey` one has not FEFF:FCFD VID/PID and has Yubikey ones
```
cmp -l pico_openpgp_pico-2.0.uf2 pico_openpgp_pico-2.0.yubikey5.uf2 | gawk '{printf "%08X %02X %02X\n", $1-1, strtonum(0$2), strtonum(0$3)}' 
00073A60 FF 50
00073A61 FE 10
00073A62 FD 07
00073A63 FC 04
```

Are you sure you are flashing it correctly?

sa2kaa, 2024-05-30:
Yes I am sure, flashing is done with internal firmware i the nano-board which is mounted as a USB-disk in my file system:
olof@olof-HP-ProBook-6450b:~$ cd Downloads/
olof@olof-HP-ProBook-6450b:~/Downloads$ ls *.uf2
pico_openpgp_pico-2.0.1050_0407.uf2  pico_openpgp_pico-2.0.yubikey5.uf2
pico_openpgp_pico-2.0.uf2
olof@olof-HP-ProBook-6450b:~/Downloads$ cp pico_openpgp_pico-2.0.yubikey5.uf2 /media/olof/RPI-RP2/
olof@olof-HP-ProBook-6450b:~/Downloads$ sudo dmesg
...

[  449.556247] usb 2-1.3: USB disconnect, device number 4
[  453.320490] usb 2-1.3: new full-speed USB device number 5 using ehci-pci
[  453.434736] usb 2-1.3: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[  453.435717] usb 2-1.3: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 5.00
[  453.435726] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  453.435731] usb 2-1.3: Product: Pico Key
[  453.435735] usb 2-1.3: Manufacturer: Pol Henarejos
[  453.435738] usb 2-1.3: SerialNumber: E6616407E32C2227

polhenarejos, 2024-05-30:
Copy that output:
```
cmp -l pico_openpgp_pico-2.0.uf2 pico_openpgp_pico-2.0.yubikey5.uf2 | gawk '{printf "%08X %02X %02X\n", $1-1, strtonum(0$2), strtonum(0$3)}' 
```

potatopplking, 2024-07-14:
Hello! This is my first post on this repository, so first of all I would like to thank @polhenarejos for a nice project :)

I also have this issue, but it seems I have managed to at least locate the source of the problem:

1. building with default VID/PID and then patching with Yubikey VID/PID
    - doesn't work, device still shows default VID/PID when connected
    - hex diff on the .uf2 files show that the VID/PID was changed compared to old binary (so it seems correct)
    - doesn't work whether I use patch script or the web patching tool (results are identical)
2. building with `-DUSB_VID=0x1050 -DUSB_PID=0x0407` 
    - does work, device shows correct (new) VID/PID

Hex diff between cases 1. and 2. shows that there is another instance of VID and PID in the binary, which is not handled by the patching script:

<pre><code>
# diff pico_openpgp.patched.uf2.hexdump pico_openpgp.yubi.uf2.hexdump
2844,2845c2844,2845
< 00010ee0  5b ba 92 b2 9b b2 d0 e7  08 1d 00 20 <b>fd fc</b> 00 00  |[.......... ....|
< 00010ef0  <b>ff fe</b> 00 00 48 1d 00 20  38 51 00 20 04 f8 00 20  |....H.. 8Q. ... |
---
> 00010ee0  5b ba 92 b2 9b b2 d0 e7  08 1d 00 20 <b>07 04</b> 00 00  |[.......... ....|
> 00010ef0  <b>50 10</b> 00 00 48 1d 00 20  38 51 00 20 04 f8 00 20  |P...H.. 8Q. ... |
</code></pre>

I am using Waveshare RP2040-One clone, so I am building with `-DPICO_BOARD=waveshare_rp2040_one`.

The reason why the patching script does not change VID/PID in this case is probably because this occurence does not conform to expected form (it is not followed by expected byte sequence).

polhenarejos, 2024-07-14:
Did you try using Pico Patcher from the web?

potatopplking, 2024-07-14:
I did, the resulting uf2 file is identical to the one patched with the script. It seems like the whole issue is that the patch does not modify all occurrences of VID/PID in the binary.

polhenarejos, 2024-07-15:
PHY is overriding patching. It'll be corrected in next release.

commit 11cb855f (2024-07-15): Fix idVendor and idProduct placeholders for Pico Patcher. Fixes #14 and #15
closed 2024-07-15 by polhenarejos