polhenarejos/pico-openpgp #15: Device Manger reports "Invalid Configuration Descriptor"

Lillythefox, 2024-06-03:
Hello, I believe the Pico Patcher is not working, the device is showing up as "Invalid Configuration Descriptor" and none of my software can detect it (GPG, kleopatra and opensc). As you see below in the event viewer logs it mentions "00FF2000" the default descriptor.

[image]

```
Device USB\VID_0000&PID_0006\5&2d531919&0&1 was configured.

Driver Name: usb.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Driver Date: 06/21/2006
Driver Version: 10.0.19041.4355
Driver Provider: Microsoft
Driver Section: BADDEVICE.Dev.NT
Driver Rank: 0xFF0000
Matching Device Id: USB\CONFIGURATION_DESCRIPTOR_VALIDATION_FAILURE
Outranked Drivers: usb.inf:USB\CONFIGURATION_DESCRIPTOR_VALIDATION_FAILURE:**00FF2000**
Device Updated: false
Parent Device: USB\ROOT_HUB30\4&30961835&0&0
```

imkuang, 2024-07-04:
Same problem with version 2.0, version 1.12 works fine.

polhenarejos, 2024-07-15:
Fixed in 11cb855

closed 2024-07-15 by polhenarejos

ItalianJack, 2024-09-01:
This appears to still be an issue with release 2.2. Compiling while setting the IDs manually works fine, however the online tool doesn't seem to change them from default.