polhenarejos/pico-openpgp #18: ykman compatible?

hkfuertes, 2024-08-12:
Hello, Just tested and patched my pico with this firmware and yubiko 4/5 vid/pid. I see this:
```bash
hkfuertes@trantor:~$ ykman list
WARNING: Failed opening device
YubiKey [OTP+FIDO+CCID] <access denied>
```
I have tested with an original Yubikey 5 Fido and ykman is properly configured (udev). Does this means that this firmware is not compatible with Yubico Software? if so, is there any GUI software to add my certificates (piv)? I saw the opensc, and I believe it worked for me, but I would love to use a GUI software (windows/linux)

Thank you!

polhenarejos, 2024-08-12:
At this moment, only `yubico-piv-tool` is supported for PIV in Pico OpenPGP.
`gpg/gpg2` is supported for the OpenPGP features and `ykman` is supported in Pico Fido for FIDO, OTP and OATH.

Unfortunately, `yubico-piv-tool` is only CLI: https://developers.yubico.com/yubico-piv-tool/

In the future, I'll bring support to `ykman` in Pico OpenPGP.

polhenarejos, 2024-08-13:
I added support to `ykman` but the OpenPGP app is broken, since Yubico does not follow the standard. PIV works smoothly. 
In case you'd need to use OpenPGP, `gpg2` is the recommended app.

Changes are available in the repo and will be released soon.

closed 2025-01-15 by polhenarejos