polhenarejos/pico-openpgp #20: Can't move ed25519/cv25519 keys to card

ViZiD, 2024-09-28:
I have keys 
```bash
/run/user/1000/gnupg/pubring.kbx
\--------------------------------
sec   ed25519/0xC2122D51CCE4FFF2 2024-09-28 [C]
      Key fingerprint = 6E22 B107 949E 5077 F405  D6BE C212 2D51 CCE4 FFF2
uid                   [ultimate] Radik Islamov <mail@vizqq.cc>
ssb   ed25519/0xF1E22078CF825EC5 2024-09-28 [S] [expires: 2026-09-28]
ssb   cv25519/0xDDBBF4D7E5B00481 2024-09-28 [E] [expires: 2026-09-28]
ssb   ed25519/0x92042AAEE5DED137 2024-09-28 [A] [expires: 2026-09-28]
```
After trying move subkeys to card, card is no longer detected in gnupg...
```bash
gpg> key 2

sec  ed25519/0xC2122D51CCE4FFF2
     created: 2024-09-28  expires: never       usage: C   
     trust: ultimate      validity: ultimate
sub  ed25519/0xF1E22078CF825EC5
     created: 2024-09-28  expires: 2026-09-28  usage: S   
ssb* cv25519/0xDDBBF4D7E5B00481
     created: 2024-09-28  expires: 2026-09-28  usage: E   
ssb  ed25519/0x92042AAEE5DED137
     created: 2024-09-28  expires: 2026-09-28  usage: A   
[ultimate] (1). Radik Islamov <mail@vizqq.cc>

gpg> keytocard
Please select where to store the key:
   (2) Encryption key
Your selection? 2

sec  ed25519/0xC2122D51CCE4FFF2
     created: 2024-09-28  expires: never       usage: C   
     trust: ultimate      validity: ultimate
sub  ed25519/0xF1E22078CF825EC5
     created: 2024-09-28  expires: 2026-09-28  usage: S   
ssb* cv25519/0xDDBBF4D7E5B00481
     created: 2024-09-28  expires: 2026-09-28  usage: E   
ssb  ed25519/0x92042AAEE5DED137
     created: 2024-09-28  expires: 2026-09-28  usage: A   
[ultimate] (1). Radik Islamov <mail@vizqq.cc>

Note: the local copy of the secret key will only be deleted with "save".
gpg> save
gpg: update failed: Card error
```
<details>
  <summary>pcsc_scan log after card broke</summary>
Scanning present readers...
Waiting for the first reader... found one
Scanning present readers...
0: Yubico YubiKey OTP+FIDO+CCID [Pico Key CCID Interface] (DE6270431F522A2B) 00 00
 
Sat Sep 28 07:13:38 2024
 Reader 0: Yubico YubiKey OTP+FIDO+CCID [Pico Key CCID Interface] (DE6270431F522A2B) 00 00
  Event number: 0
  Card state: Card inserted, 
  ATR: 3B DA 18 FF 81 B1 FE 75 1F 03 00 31 F5 73 C0 01 60 00 90 00 1C

ATR: 3B DA 18 FF 81 B1 FE 75 1F 03 00 31 F5 73 C0 01 60 00 90 00 1C
+ TS = 3B --> Direct Convention
+ T0 = DA, Y(1): 1101, K: 10 (historical bytes)
  TA(1) = 18 --> Fi=372, Di=12, 31 cycles/ETU
    129032 bits/s at 4 MHz, fMax for Fi = 5 MHz => 161290 bits/s
  TC(1) = FF --> Extra guard time: 255 (special value)
  TD(1) = 81 --> Y(i+1) = 1000, Protocol T = 1 
\-----
  TD(2) = B1 --> Y(i+1) = 1011, Protocol T = 1 
\-----
  TA(3) = FE --> IFSC: 254
  TB(3) = 75 --> Block Waiting Integer: 7 - Character Waiting Integer: 5
  TD(3) = 1F --> Y(i+1) = 0001, Protocol T = 15 - Global interface bytes following 
\-----
  TA(4) = 03 --> Clock stop: not supported - Class accepted by the card: (3G) A 5V B 3V 
+ Historical bytes: 00 31 F5 73 C0 01 60 00 90 00
  Category indicator byte: 00 (compact TLV data object)
    Tag: 3, len: 1 (card service data byte)
      Card service data byte: F5
        \- Application selection: by full DF name
        \- Application selection: by partial DF name
        \- BER-TLV data objects available in EF.DIR
        \- BER-TLV data objects available in EF.ATR
        \- EF.DIR and EF.ATR access services: by GET DATA command
        \- Card without MF
    Tag: 7, len: 3 (card capabilities)
      Selection methods: C0
        \- DF selection by full DF name
        \- DF selection by partial DF name
      Data coding byte: 01
        \- Behaviour of write functions: one-time write
        \- Value 'FF' for the first byte of BER-TLV tag fields: invalid
        \- Data unit in quartets: 2
      Command chaining, length fields and logical channels: 60
        \- Extended Lc and Le fields
        \- RFU (should not happen)
        \- Logical channel number assignment: No logical channel
        \- Maximum number of logical channels: 1
    Mandatory status indicator (3 last bytes)
      LCS (life card cycle): 00 (No information given)
      SW: 9000 (Normal processing.)
+ TCK = 1C (correct checksum)

Possibly identified card (using /nix/store/qd5x13g2kqlaj3rf5d6rvpdnbym3x9s1-pcsc-tools-1.7.2/share/pcsc/smartcard_list.txt):
3B DA 18 FF 81 B1 FE 75 1F 03 00 31 F5 73 C0 01 60 00 90 00 1C
	OpenPGP Card V3
</details>

I use waveshare rp2040 one, firmware version 2.2

polhenarejos, 2024-09-28:
Please provide steps to reproduce it, including the generation of ed25519 key in localhost.

ViZiD, 2024-09-28:
```bash 
export IDENTITY="johh locke <johh@locke.me>"
export EXPIRATION=2y

gpg --pinentry-mode=loopback --quick-generate-key "$IDENTITY" ed25519 cert never

export KEYFP=$(gpg -k --with-colons "$IDENTITY" | awk -F: '/^fpr:/ { print $10; exit }')

gpg --pinentry-mode=loopback --quick-add-key $KEYFP ed25519 sign $EXPIRATION
gpg --pinentry-mode=loopback --quick-add-key $KEYFP cv25519 encr $EXPIRATION
gpg --pinentry-mode=loopback --quick-add-key $KEYFP ed25519 auth $EXPIRATION

gpg --edit-key $KEYFP 

Secret key is available.

sec  ed25519/0x54C046F05B051A89
     created: 2024-09-28  expires: never       usage: C   
     trust: ultimate      validity: ultimate
ssb  ed25519/0xE7FA1A03722683A8
     created: 2024-09-28  expires: 2026-09-28  usage: S   
ssb  cv25519/0xFC452AADEE3DC41F
     created: 2024-09-28  expires: 2026-09-28  usage: E   
ssb  ed25519/0x4D7D7CDA4128AC7E
     created: 2024-09-28  expires: 2026-09-28  usage: A   
[ultimate] (1). johh locke <johh@locke.me>

gpg> key 1

sec  ed25519/0x54C046F05B051A89
     created: 2024-09-28  expires: never       usage: C   
     trust: ultimate      validity: ultimate
ssb* ed25519/0xE7FA1A03722683A8
     created: 2024-09-28  expires: 2026-09-28  usage: S   
ssb  cv25519/0xFC452AADEE3DC41F
     created: 2024-09-28  expires: 2026-09-28  usage: E   
ssb  ed25519/0x4D7D7CDA4128AC7E
     created: 2024-09-28  expires: 2026-09-28  usage: A   
[ultimate] (1). johh locke <johh@locke.me>

gpg> keytocard
Please select where to store the key:
   (1) Signature key
   (3) Authentication key
Your selection? 1

# move cv25519 key
gpg --edit-key $KEYFP 
Secret key is available.

sec  ed25519/0x54C046F05B051A89
     created: 2024-09-28  expires: never       usage: C   
     trust: ultimate      validity: ultimate
sub  ed25519/0xE7FA1A03722683A8
     created: 2024-09-28  expires: 2026-09-28  usage: S   
ssb  cv25519/0xFC452AADEE3DC41F
     created: 2024-09-28  expires: 2026-09-28  usage: E   
ssb  ed25519/0x4D7D7CDA4128AC7E
     created: 2024-09-28  expires: 2026-09-28  usage: A   
[ultimate] (1). johh locke <johh@locke.me>

gpg> key 2

sec  ed25519/0x54C046F05B051A89
     created: 2024-09-28  expires: never       usage: C   
     trust: ultimate      validity: ultimate
sub  ed25519/0xE7FA1A03722683A8
     created: 2024-09-28  expires: 2026-09-28  usage: S   
ssb* cv25519/0xFC452AADEE3DC41F
     created: 2024-09-28  expires: 2026-09-28  usage: E   
ssb  ed25519/0x4D7D7CDA4128AC7E
     created: 2024-09-28  expires: 2026-09-28  usage: A   
[ultimate] (1). johh locke <johh@locke.me>

gpg> keytocard
Please select where to store the key:
   (2) Encryption key
Your selection? 2
```

polhenarejos, 2024-09-30:
Are you using the EdDSA branch?

ViZiD, 2024-09-30:
> Are you using the EdDSA branch?

Hi, no, i use 2.2 version from release page

ViZiD, 2024-10-04:
> Are you using the EdDSA branch?

I try eddsa branch, and it work ;-)

closed 2025-01-15 by polhenarejos

ViZiD, 2025-05-11:
I have the same problem, and have had it for quite some time, since about version 2.2
I don't use gpg to change the password, instead I use openpgp-card-tools https://codeberg.org/openpgp-card/openpgp-card-tools, it changes the pin perfectly there.
I experimented a bit and found out that if you change the reset pin first, then the user pin and admin pin will change fine after that

```sh
#!/usr/bin/env bash

ADMIN_PASS="12345678"
PIN_PASS="123456"

function change_user_and_admin_pin {
	gpg --command-fd=0 --pinentry-mode=loopback --edit-card <<EOF
admin
passwd
1
$PIN_PASS
654321
654321
3
$ADMIN_PASS
87654321
87654321
q
q
EOF
}

function change_reset_pin {
	gpg --command-fd=0 --pinentry-mode=loopback --edit-card <<EOF
admin
passwd
4
$ADMIN_PASS
87654321
87654321
q
q
EOF
}

change_user_and_admin_pin
change_reset_pin
change_user_and_admin_pin
```
<details>
<summary>After that you can see that user pin and admin pin has been successfully changed.</summary>
Reader ...........: Yubico YubiKey CCID [Pico Key HID Interface] (A9B20E0FC513E195) 00 00
Application ID ...: D276000124010304FFFEA9B20E0F0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unmanaged S/N range
Serial number ....: A9B20E0F
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: not forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
UIF setting ......: Sign=off Decrypt=off Auth=off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]

Admin commands are allowed

gpg: OpenPGP card no. D276000124010304FFFEA9B20E0F0000 detected

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Error changing the PIN: Card error

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Error changing the PIN: Card error

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Reader ...........: Yubico YubiKey CCID [Pico Key HID Interface] (A9B20E0FC513E195) 00 00
Application ID ...: D276000124010304FFFEA9B20E0F0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unmanaged S/N range
Serial number ....: A9B20E0F
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: not forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
UIF setting ......: Sign=off Decrypt=off Auth=off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]

Admin commands are allowed

gpg: OpenPGP card no. D276000124010304FFFEA9B20E0F0000 detected

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Reset Code set.

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Reader ...........: Yubico YubiKey CCID [Pico Key HID Interface] (A9B20E0FC513E195) 00 00
Application ID ...: D276000124010304FFFEA9B20E0F0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unmanaged S/N range
Serial number ....: A9B20E0F
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: not forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
UIF setting ......: Sign=off Decrypt=off Auth=off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]

Admin commands are allowed

gpg: OpenPGP card no. D276000124010304FFFEA9B20E0F0000 detected

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

PIN changed.

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

PIN changed.

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit
</details>