polhenarejos/pico-openpgp #21: Generic rp2040 with usb, which release

nmwael, 2024-10-19:
if i bought a Generic rp2040 usb dongle, which release should i choose?

polhenarejos, 2024-10-19:
The most important figure is the flash size. Usually these generic boards are based on waveshare zero

nmwael, 2024-10-19:
Thanks i will try then

nmwael, 2024-11-23:
So after I've flashed sucessfully. What steps should then be taken? OpenSC seems to be able to read, Kleopatra also.. However I cannot do any operations since I do not know the pin? I've used pico-commissioner to initialize?

polhenarejos, 2024-11-23:
Commissioner is useful to change default VID PID, initialize or change some parameters. Id OpenSC detects it, gpg2 should too. 
Default pins for user and admin are 123456 and 12345678.

nmwael, 2024-11-26:
thanks, can you tell how I activate 
```
User interaction flag (UIF) to enable/disable press-to-confirm button
```

closed 2025-01-15 by polhenarejos