polhenarejos/pico-openpgp #24: Pico commissionner not working

Tartasprint, 2024-12-06:
Hello, first of all thank you for your amazing work !
I was able to set up a FIDO key on Pico2 following the instructions on picokeys.com:
1) download the firmware and write it from the getting started page
2) use the commisionner
I tried to follow the same steps with the OpenGPG and HSM firmware, but the commissionner/browser/OS is not detecting the device properly. I am on Windows, and I tried both Chrome and Firefox (the latter only with WebAuthn).

Kleopatra (a GUI for gpg on windows) successfully detects the "smart card". I tried to use several trivial PINs, before finding out that the default PIN from the HSM firmware worked also on OpenGPG firmware. The behaviour is different, Kleopatra prompts for changes instead of saying "Incorrect PIN". At  the last step I get "Conditions d'utilisations non-satisfaites" which translates to "Usage condition are not met/satisfied".

Did I miss a step between writing the firmware and commissionning it ? I did try to unplug and replug, even if it is not supposed be needed.

polhenarejos, 2025-01-15:
Commissioner uses WebUSB and it requires permission from your browser. Check whether you granted the proper access.

nmwael, 2025-02-25:
I seem to have the same problem, on pico_openpgp_waveshare_rp2040_plus_16mb-3.2 , version 3.4 is not booting on my board.

MuRongPIG, 2025-03-22:
The same problem .I was using esp32-s3 and after commissioning the vid/pid turned 0000:0000 no matter what the preset vendor is.  @polhenarejos Could you help check where the problem is? thank you! <3 (after flashing the vid/pid is normal FEFF:FCFD)

MuRongPIG, 2025-03-22:
btw ,Can I change the vid/pid without the commissioner for esp32-s3? (like pico-fido)(because I dont know how to build for esp32)

polhenarejos, 2025-03-22:
You need to use development firmware.

petrm, 2025-03-27:
> The same problem .I was using esp32-s3 and after commissioning the vid/pid turned 0000:0000 no matter what the preset vendor is. [@polhenarejos](https://github.com/polhenarejos) Could you help check where the problem is? thank you! <3 (after flashing the vid/pid is normal FEFF:FCFD)

I am facing the same issue on rp2040, after using the commissioner the vid:pid is 0000:0000. Works fine with pico fido.

mtorromeo, 2025-04-01:
The issue is not limited to the commissioner, actually the problem is that the firmware is not working correctly at all.

I tried compiling it by myself and the only difference was that the LED did not work at all (I am sure I am missing something), but the device was still recognized but with the wrong VID:PID `0000:0000`, and this is the symptom of the real issue, since the default VID:PID is supposed to be `FEFF:FCFD` even when not using the commissioner.

So I tried to flash the older stable releases, one at a time, hoping to find one that worked, and I did.

The first working release was 2.0, which got me the blinking green led and the proper lsusb listing:
```
Bus 003 Device 016: ID feff:fcfd Pol Henarejos Pico Key
```

Something after that changed, and any release from 3.0 onward results in red LED and `0000:0000`.

mtorromeo, 2025-04-01:
Actually, a small correction: the first working release is v2.2 not v2.0. Sorry about that.

mtorromeo, 2025-04-01:
I bisected commits from v2.2 to v3.0 and when I found the commit that updated pico-keys-sdk I started bisecting those commits and found the culprit in polhenarejos/pico-keys-sdk@0638409ff82cb1327283bd6795bc26b68b05c936

Starting from the commit the vid:pid became `0000:0000`.

Before that the device was recognized used `1234:5678`, even when led became solid red, or blinking red before that, and solid even cyan at one point.

lsusb output at commit polhenarejos/pico-keys-sdk@802df9e70534e8565d9e9547ed9f8bd2dac762ea:
```
Bus 003 Device 085: ID 1234:5678 Brain Actuated Technologies Pico Key
```

lsusb output at commit polhenarejos/pico-keys-sdk@0638409ff82cb1327283bd6795bc26b68b05c936:
```
Bus 003 Device 087: ID 0000:0000 Pol Henarejos Pico Key
```

It's interesting that the device manufactured also changed between commits of pico-keys-sdk.

damko, 2025-12-28:
> > btw the commissioner is no more online
> 
> How did you manage to figure that out after at least 7 threads on exactly this topic?

Apologies - I didn't mean to be rude or to add pressure. I'm new to this project and I haven't read all the issues.

I simply thought to drop a line in case it went unnoticed. Behind my post there was nothing but the will to help.

wlatendresse, 2025-12-28:
> 
> Apologies - I didn't mean to be rude or to add pressure. I'm new to this project and I haven't read all the issues.

And yet, the problem is growing that important information on the internet gets lost in the noise of trivialities.

Read before you maybe post. Especially since every interested person is notified of every trivial matter.

polhenarejos, 2026-08-20:
Closed due to inactivity. Reopen if needed.

closed 2026-08-20 by polhenarejos