polhenarejos/pico-openpgp #25: Consider overclocking the Raspberry Pi Pico microcontroller

c0m1c5an5, 2024-12-14:
This is a question for the development of this project for Raspberry Pi pico. Has overclocking been considered?
There has been plenty of information about overclocking available in this [forum thread](https://forums.raspberrypi.com/viewtopic.php?t=301902).
I believe using overclocking can significantly improve the operating time, making slower algorithms more viable.
If there is no reason not to use overclocking, then it would be a nice feature to have.

PS. If this is the feature to be approved, but there is no one who can allocate their time for development, I could take a stab at it. I have no experience with Raspberry Pi pico though.

polhenarejos, 2024-12-14:
The target of Pico Keys is not being the fastest boards in the market, but having a robust tiny space to allocate private and secret keys, with the highest security as possible given the hardware. Overclocking reduces significantly the life span of your board and thus, losing all your secure material prematurely. Of course, it will happen with any device, with our without overclock, but doing it for all the public is a no-deal when we talk about secure devices and providing a minimum reliability. Fortunately, Pico Keys is an open-source and open-hardware project, so everyone can overclock their boards if they want at their own risk.

c0m1c5an5, 2024-12-14:
Seems logical to me. Still nice to have this discussion to provide the reasoning to others who have the same question.

closed 2024-12-14 by c0m1c5an5