polhenarejos/pico-openpgp #26: "gpg: KEYTOCARD failed: General error" with ed25519 key type on branch eddsa

13pgeiser, 2025-01-04:
Hello Pol,

First of all, thank you for sharing this work.

I'm trying to send ed25519 keys to use my tiny2530 board as a smartcard with gnupg. I've tried many different combinations (of VIDPID and boards: pico,  tiny2530)  without success. Even if it looks similar to https://github.com/polhenarejos/pico-openpgp/issues/20, it seems it's not the same problem.

Here is the script I use to rebuild the firmware and test it. 

**WARNING: the script deletes ".gnupg" folder.**

```
#!/bin/bash
set -ex
# Using debian 12
cat /etc/os-release
sudo apt install -y cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib  opensc gnupg
# Clone both pico-sdk & pico-openpgp
git clone https://github.com/raspberrypi/pico-sdk.git --branch 2.0.0 --recurse-submodules
git clone https://github.com/polhenarejos/pico-openpgp.git --branch eddsa --recurse-submodules
# Build
mkdir -p pico-build
cd pico-build
cmake -DPICO_BOARD=pimoroni_tiny2350 -DVIDPID=Gnuk -DPICO_SDK_PATH="../pico-sdk/" ../pico-openpgp
make -j"$(nproc)"
# Copy to board and wait a bit.
cp pico_openpgp.uf2 "/media/$USER/RP2350/"
sleep 30
# WARNING deletes actual gnupg installation.
rm -rf ~/.gnupg
gpg --card-status
# Create new keys
export IDENTITY="pico openpgp<pico@openpgp.me>"
export CERTIFY_PASS="test"
echo "$CERTIFY_PASS" | gpg --batch --passphrase-fd 0 --quick-generate-key "$IDENTITY" ed25519 cert never
export KEYFP=$(gpg -k --with-colons "$IDENTITY" | awk -F: '/^fpr:/ { print $10; exit }')
echo "$CERTIFY_PASS" | gpg --batch --pinentry-mode=loopback --passphrase-fd 0 --quick-add-key "$KEYFP" ed25519 sign 1y
gpg -K
# Put key on card
export KEYID=$(gpg -k --with-colons "$IDENTITY" | awk -F: '/^pub:/ { print $5; exit }')
gpg --command-fd=0 --pinentry-mode=loopback --edit-key $KEYID <<EOF
key 1
keytocard
1
$CERTIFY_PASS
EOF

```

Attached, the full log: [pico-log.txt](https://github.com/user-attachments/files/18307500/pico-log.txt)

If I try to send a RSA key with this firmware, I get same failure.

If I try to send RSA keys with main branches, it seems to work

If I send a ed25519 with the main branches, I get "gpg: KEYTOCARD failed: Not supported".

polhenarejos, 2025-01-04:
I think there's a problem with your code. You are missing the Admin PIN after the passphrase:

```
gpg --command-fd=0 --pinentry-mode=loopback --edit-key $KEYID <<EOF
key 1
keytocard
1
$CERTIFY_PASS
$ADMIN_PASS
$ADMIN_PASS
EOF

By default admin pass is `12345678`.

13pgeiser, 2025-01-05:
In fact, the problem was happening before asking for Admin Pin but you're right, these 2 were missing!

I don't know precisely what I did to put the board in this state but downloading [flash_nuke.uf2](https://datasheets.raspberrypi.com/soft/flash_nuke.uf2) to fully erase the flash solved the issue.

Sorry for the noise and many thanks for sharing this code!

BR,

Pascal.

closed 2025-01-05 by 13pgeiser