polhenarejos/pico-openpgp #32: can not change openpgp password

CGsama, 2025-04-22:
pico_openpgp_pico2-3.6-eddsa1

either pin or admin pin can not be changed, ok on pico w but not working on pico 2

Error changing the PIN: Card error

polhenarejos, 2025-04-22:
Did you nuke it first?

CGsama, 2025-04-22:
> Did you nuke it first?

i tried on two brand new boards, let me try nuke

CGsama, 2025-04-22:
tried with pico_nuke_pico2-1.4.uf2 not working, still cant change pin

ViZiD, 2025-05-11:
I have the same problem, and have had it for quite some time, since about version 2.2
I don't use gpg to change the password, instead I use openpgp-card-tools https://codeberg.org/openpgp-card/openpgp-card-tools, it changes the pin perfectly there.
I experimented a bit and found out that if you change the reset pin first, then the user pin and admin pin will change fine after that

```sh
#!/usr/bin/env bash

ADMIN_PASS="12345678"
PIN_PASS="123456"

function change_user_and_admin_pin {
	gpg --command-fd=0 --pinentry-mode=loopback --edit-card <<EOF
admin
passwd
1
$PIN_PASS
654321
654321
3
$ADMIN_PASS
87654321
87654321
q
q
EOF
}

function change_reset_pin {
	gpg --command-fd=0 --pinentry-mode=loopback --edit-card <<EOF
admin
passwd
4
$ADMIN_PASS
87654321
87654321
q
q
EOF
}

change_user_and_admin_pin
change_reset_pin
change_user_and_admin_pin
```
<details>
<summary>After that you can see that user pin and admin pin has been successfully changed.</summary>
Reader ...........: Yubico YubiKey CCID [Pico Key HID Interface] (A9B20E0FC513E195) 00 00
Application ID ...: D276000124010304FFFEA9B20E0F0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unmanaged S/N range
Serial number ....: A9B20E0F
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: not forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
UIF setting ......: Sign=off Decrypt=off Auth=off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]

Admin commands are allowed

gpg: OpenPGP card no. D276000124010304FFFEA9B20E0F0000 detected

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Error changing the PIN: Card error

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Error changing the PIN: Card error

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Reader ...........: Yubico YubiKey CCID [Pico Key HID Interface] (A9B20E0FC513E195) 00 00
Application ID ...: D276000124010304FFFEA9B20E0F0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unmanaged S/N range
Serial number ....: A9B20E0F
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: not forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
UIF setting ......: Sign=off Decrypt=off Auth=off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]

Admin commands are allowed

gpg: OpenPGP card no. D276000124010304FFFEA9B20E0F0000 detected

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Reset Code set.

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

Reader ...........: Yubico YubiKey CCID [Pico Key HID Interface] (A9B20E0FC513E195) 00 00
Application ID ...: D276000124010304FFFEA9B20E0F0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unmanaged S/N range
Serial number ....: A9B20E0F
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: not forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
UIF setting ......: Sign=off Decrypt=off Auth=off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]

Admin commands are allowed

gpg: OpenPGP card no. D276000124010304FFFEA9B20E0F0000 detected

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

PIN changed.

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit

PIN changed.

1 - change PIN
2 - unblock PIN
3 - change Admin PIN
4 - set the Reset Code
Q - quit
</details>

polhenarejos, 2025-05-12:
Seems a problem of initialization.

polhenarejos, 2025-05-12:
It was a problem of the order of loading DEK and PIN check.

commit fccc48de (2025-05-12): Fix change PIN when no PIN was introduced before.

Fixes #32
closed 2025-05-12 by polhenarejos