polhenarejos/pico-openpgp #35: Unable to change usb vidpid after initial comissioning

axtloss, 2025-06-24:
i accidentally commissioned my pico-opengpg with a yubikey vidpid, even though i needed the gnupg one and I cannot change it anymore.
To change it i tried:
- patching the firmware to contain the vidpid
- using the web commissioner, webusb doesn't see the pico, webauthn doesn't either
- flashing the pico-fido firmware and using pico-fido-tool.py but it errors with connection timed out
- using pico-nuke to remove everything and flashing it again, but the vidpid persists

I am using the pico 2 firmware with [this board](https://www.aliexpress.com/item/1005008640853778.html), i have another one with pico-fido which works fine (aside from leds not working)

polhenarejos, 2025-08-11:
Yubikey vidpid is blocked from WebUSB, so it cannot be recommissioned and webauthn is for Pico Fido.
Try with `picotool erase -a` in BOOTSEL mode.

polhenarejos, 2026-08-20:
Closed due to inactivity. Reopen if needed.

closed 2026-08-20 by polhenarejos