polhenarejos/pico-openpgp #36: The management key of piv cannot be changed

Orangeemen, 2025-07-06:
I tried to use Yubico Authenticator to change my management key, but after my change, the piv function of my device was no longer available. The error message is as follows.

RpcResponse.error(status: invalid-command, message:
Invalid parameters: 255 is not a valid TOUCH_POLICY,
body: 0)

Device: waveshare_rp2350_one
Firmware: pico_fido2_waveshare_rp2350_one-6.6

[image]

polhenarejos, 2025-07-06:
Please put the steps to reproduce it.

Orangeemen, 2025-07-07:
> Please put the steps to reproduce it.

 Yes ， I reproduced it. It works again. So is that means it is not available to change management key on pico

Young-Lord, 2025-07-31:
还请附上中文原文，或者用个靠谱的翻译软件。我不理解你想表达的意思。
Please also attach the original Chinese text or use a reliable translation software. I don't understand what you mean.